fan-ziqi/rl_sar #3: Abandoned (core has been dumped)

hanzhi0410, 2024-04-26:
Hello author, when using your deployment plan, you often encounter the following problems by chance
#########
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
##########

May I ask if you have encountered any good solutions during use

fan-ziqi, 2024-04-26:
Please use gdb to debug the program and reply with specific errors

hanzhi0410, 2024-05-21:
Thank you for your reply. I added torch:: autograd:: GradMode:: set_enabled (false); Solved the problem.

closed 2024-05-21 by hanzhi0410

fan-ziqi, 2024-05-21:
Maybe you didn't pull the latest version, this issue has been fixed in 81fa87035a512f2eeaac079bd35a256ffef7d413

hanzhi0410, 2024-05-21:
It may be an issue with my model, and I also need to add torch:: autograd:: GradMode:: set_enabled (false) in torch:: Tensor RL_Sim:: Forward(); Like this:
Torch:: Tensor RL_Sim:: Forward()
{
Torch:: autograd:: GradMode:: set_enabled (false);
Torch:: Tensor action=this ->model. forward ({this ->unobsbuffer_tensor}). toTensor();
This ->obs. actions=action;
Torch:: Tensor clamped=Torch:: clamp (action, - this ->params. clip.actions, this ->params. clip.actions);
Return clamped;
}