fan-ziqi/rl_sar #4: How to choose data for Cyberdog training?

RubSevian, 2024-05-09:
I want to repeat your success and possibly improve walking, and for this I really need information about where you conducted and what you based your Cyberdog training on.Please

fan-ziqi, 2024-05-12:
I use [legged_gym](https://github.com/leggedrobotics/legged_gym) as the base framework, and the model is based on a modified version of cyberdog1_description with parameters provided by the official repository [cyberdog1_description](https://github.com/fan-ziqi/cyberdog1_description). The environment setup uses the script [rl_docker](https://github.com/fan-ziqi/rl_docker).

However, after testing, it was found that using Kp=20 and Kd=0.5 does not enable walking on the real dog. Therefore, it is recommended to use the actuator_network instead of PD for training.

RubSevian, 2024-05-31:
Hello, I have assembled your rl_docker.I would like to learn more about the Cyberdog robot training process, can you tell me more about it?

fan-ziqi, 2024-06-01:
Please join Discord group in rl_sar README

closed 2024-06-01 by fan-ziqi