fan-ziqi/rl_sar #11: I don't quite understand the compilation and running parts.

wangxudong1998, 2024-08-13:
I don't quite understand the **compilation** and **running** parts.

Where can I customize the following two functions in my code? There is indeed compute_observable() in my code.
And then compile it in the root directory of my code?
```
torch::Tensor forward() override;
torch::Tensor compute_observation() override;
```

In the running section, I encountered the following error:
```
$ source devel/setup.bash
bash: devel/setup.bash: No such file or directory
```
```
$ ros2 launch rl_sar gazebo_gr1t1.launch
Package 'rl_sar' not found: "package 'rl_sar' not found, searching: ['/opt/ros/humble']"
```
How should I solve it? Thank you.

fan-ziqi, 2024-08-13:
This project uses ROS1(noetic), however you are using ROS2(humble).

fan-ziqi, 2024-08-13:
https://github.com/fan-ziqi/rl_sar/blob/e42d926bf3d3ad0299c6c2f58b32f1423d632cff/src/rl_sar/src/rl_real_a1.cpp#L173-L202

or

https://github.com/fan-ziqi/rl_sar/blob/e42d926bf3d3ad0299c6c2f58b32f1423d632cff/src/rl_sar/src/rl_sim.cpp#L247-L281

closed 2024-08-13 by fan-ziqi