fan-ziqi/rl_sar #12: PytorchStreamReader failed locating file constants.pkl: file not found

kunlunxu1, 2024-08-14:
我使用iscaacsim训练的文件中没有constants.pkl
[image]
[image]
请问大佬的policy是怎么训练的

fan-ziqi, 2024-08-14:
本项目不需要constants.pkl，训练过程参考https://github.com/fan-ziqi/robot_lab

fan-ziqi, 2024-08-14:
你有没有导出为jit？
https://github.com/fan-ziqi/robot_lab/blob/14d8b9d18354bccd9dfccb198f8d11ea9303887d/scripts/rsl_rl/play.py#L124-L129

kunlunxu1, 2024-08-14:
> 你有没有导出为jit？
> 
> https://github.com/fan-ziqi/robot_lab/blob/14d8b9d18354bccd9dfccb198f8d11ea9303887d/scripts/rsl_rl/play.py#L124-L129
> 
> ```python
>     export_policy_as_jit(
>         actor_critic=ppo_runner.alg.actor_critic,
>         normalizer=ppo_runner.obs_normalizer,
>         path=export_model_dir,
>         filename="policy.pt",
>     )
> ```

是没有导出为jit，我去试一下，谢谢大佬

closed 2024-08-14 by fan-ziqi

popocheng, 2025-02-12:
大佬们问题解决了吗，我这边有导出为jit，还是报这个错误