fan-ziqi/rl_sar #15: RuntimeError: PytorchStreamReader failed locating file constants.pkl: file not found

normlpl, 2024-08-19:
通过cyberdog_gym得到的a1模型model_3000.pt无法在rl_sqr项目中使用，报错信息为RuntimeError: PytorchStreamReader failed locating file constants.pkl: file not found。
同时使用导出的policy_1.pt文件同样报错：RuntimeError: mat1 and mat2 shapes cannot be multiplied (1x45 and 48x512)
求解！！

fan-ziqi, 2024-08-19:
For 1st question, check issue https://github.com/fan-ziqi/rl_sar/issues/12

For 2nd question, `mat1 and mat2 shapes cannot be multiplied (1x45 and 48x512)`

closed 2024-08-19 by fan-ziqi

normlpl, 2024-08-19:
导出模型后，使用的policy_1.pt报错：RuntimeError: mat1 and mat2 shapes cannot be multiplied (1x270 and 48x512)，这是什么原因

normlpl, 2024-08-19:
而且导出的模型格式与a1_isaacgym中给的示例模型格式不一样

homemoon-cpu, 2025-02-07:
你好，我也遇到了这个问题，看样子是训练时的输入层维度是48，但推理的模型的输入变成1X270了，请问你找到原因了吗

> 导出模型后，使用的policy_1.pt报错：RuntimeError: mat1 and mat2 shapes cannot be multiplied (1x270 and 48x512)，这是什么原因