fan-ziqi/rl_sar #19: gazebo did not successfully load the robot into the scene

gaoboyi, 2024-08-30:
roslaunch rl_sar gazebo_gr1t1_isaacgym.launch    ## terminal

... logging to /home/gaoboyi/.ros/log/fd5786fc-66a2-11ef-a7cd-0d2f491c0acd/roslaunch-gaoboyi-GeekPro-17IRR-32483.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gaoboyi-GeekPro-17IRR:33725/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /gazebo_model_name: gr1t1_gazebo
 * /gr1t1_gazebo/joint_state_controller/publish_rate: 1000
 * /gr1t1_gazebo/joint_state_controller/type: joint_state_contr...
 * /gr1t1_gazebo/l_ankle_pitch_controller/joint: l_ankle_pitch
 * /gr1t1_gazebo/l_ankle_pitch_controller/pid/d: 1.5
 * /gr1t1_gazebo/l_ankle_pitch_controller/pid/i: 0.0
 * /gr1t1_gazebo/l_ankle_pitch_controller/pid/p: 15.3
 * /gr1t1_gazebo/l_ankle_pitch_controller/type: robot_joint_contr...
 * /gr1t1_gazebo/l_hip_pitch_controller/joint: l_hip_pitch
 * /gr1t1_gazebo/l_hip_pitch_controller/pid/d: 11.4
 * /gr1t1_gazebo/l_hip_pitch_controller/pid/i: 0.0
 * /gr1t1_gazebo/l_hip_pitch_controller/pid/p: 114.0
 * /gr1t1_gazebo/l_hip_pitch_controller/type: robot_joint_contr...
 * /gr1t1_gazebo/l_hip_roll_controller/joint: l_hip_roll
 * /gr1t1_gazebo/l_hip_roll_controller/pid/d: 5.7
 * /gr1t1_gazebo/l_hip_roll_controller/pid/i: 0.0
 * /gr1t1_gazebo/l_hip_roll_controller/pid/p: 57.0
 * /gr1t1_gazebo/l_hip_roll_controller/type: robot_joint_contr...
 * /gr1t1_gazebo/l_hip_yaw_controller/joint: l_hip_yaw
 * /gr1t1_gazebo/l_hip_yaw_controller/pid/d: 4.3
 * /gr1t1_gazebo/l_hip_yaw_controller/pid/i: 0.0
 * /gr1t1_gazebo/l_hip_yaw_controller/pid/p: 43.0
 * /gr1t1_gazebo/l_hip_yaw_controller/type: robot_joint_contr...
 * /gr1t1_gazebo/l_knee_pitch_controller/joint: l_knee_pitch
 * /gr1t1_gazebo/l_knee_pitch_controller/pid/d: 11.4
 * /gr1t1_gazebo/l_knee_pitch_controller/pid/i: 0.0
 * /gr1t1_gazebo/l_knee_pitch_controller/pid/p: 114.0
 * /gr1t1_gazebo/l_knee_pitch_controller/type: robot_joint_contr...
 * /gr1t1_gazebo/r_ankle_pitch_controller/joint: r_ankle_pitch
 * /gr1t1_gazebo/r_ankle_pitch_controller/pid/d: 1.5
 * /gr1t1_gazebo/r_ankle_pitch_controller/pid/i: 0.0
 * /gr1t1_gazebo/r_ankle_pitch_controller/pid/p: 15.3
 * /gr1t1_gazebo/r_ankle_pitch_controller/type: robot_joint_contr...
 * /gr1t1_gazebo/r_hip_pitch_controller/joint: r_hip_pitch
 * /gr1t1_gazebo/r_hip_pitch_controller/pid/d: 11.4
 * /gr1t1_gazebo/r_hip_pitch_controller/pid/i: 0.0
 * /gr1t1_gazebo/r_hip_pitch_controller/pid/p: 114.0
 * /gr1t1_gazebo/r_hip_pitch_controller/type: robot_joint_contr...
 * /gr1t1_gazebo/r_hip_roll_controller/joint: r_hip_roll
 * /gr1t1_gazebo/r_hip_roll_controller/pid/d: 5.7
 * /gr1t1_gazebo/r_hip_roll_controller/pid/i: 0.0
 * /gr1t1_gazebo/r_hip_roll_controller/pid/p: 57.0
 * /gr1t1_gazebo/r_hip_roll_controller/type: robot_joint_contr...
 * /gr1t1_gazebo/r_hip_yaw_controller/joint: r_hip_yaw
 * /gr1t1_gazebo/r_hip_yaw_controller/pid/d: 4.3
 * /gr1t1_gazebo/r_hip_yaw_controller/pid/i: 0.0
 * /gr1t1_gazebo/r_hip_yaw_controller/pid/p: 43.0
 * /gr1t1_gazebo/r_hip_yaw_controller/type: robot_joint_contr...
 * /gr1t1_gazebo/r_knee_pitch_controller/joint: r_knee_pitch
 * /gr1t1_gazebo/r_knee_pitch_controller/pid/d: 11.4
 * /gr1t1_gazebo/r_knee_pitch_controller/pid/i: 0.0
 * /gr1t1_gazebo/r_knee_pitch_controller/pid/p: 114.0
 * /gr1t1_gazebo/r_knee_pitch_controller/type: robot_joint_contr...
 * /robot_description: <?xml version="1....
 * /robot_name: gr1t1_isaacgym
 * /ros_namespace: /gr1t1_gazebo/
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    urdf_spawner (gazebo_ros/spawn_model)
  /gr1t1_gazebo/
    controller_spawner (controller_manager/spawner)

auto-starting new master
process[master]: started with pid [32491]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fd5786fc-66a2-11ef-a7cd-0d2f491c0acd
process[rosout-1]: started with pid [32505]
started core service [/rosout]
process[gazebo-2]: started with pid [32512]
process[gazebo_gui-3]: started with pid [32516]
process[urdf_spawner-4]: started with pid [32521]
process[gr1t1_gazebo/controller_spawner-5]: started with pid [32523]
process[robot_state_publisher-6]: started with pid [32524]
[ INFO] [1725003570.011324150]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1725003570.012185098]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1725003570.046789599]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1725003570.047413803]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1725003570.681900002]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1725003570.691307289]: Physics dynamic reconfigure ready.
[image]

closed 2024-10-12 by fan-ziqi

Lockedb, 2024-10-18:
Excuse me, has it been resolved? I also encountered the problem of roslaunch not displaying the robot model after opening Gazebo. I would like to ask how it was resolved

funnylike, 2024-10-19:
> Excuse me, has it been resolved? I also encountered the problem of roslaunch not displaying the robot model after opening Gazebo. I would like to ask how it was resolved

i have solved it，i solved it by logging out of the virtual environment.
conda deactivate