fan-ziqi/rl_sar #24: Simplified readme dependency installation

legubiao, 2024-10-08:
Hello Fan,

Thank you for your great work.

I noticed that in the README file, you used `git clone` and `make install` to install the `yaml-cpp` and `lcm` libraries. As I tested, both of them can be installed using the apt command:

```bash
sudo apt-get install liblcm-dev libyaml-cpp-dev
```

I think this process will be easier for beginners. As I tested on my Ubuntu 20.04, I didn’t encounter any problems; both the Gazebo simulation and control program worked smoothly.

Does this look good to you?

fan-ziqi, 2024-10-08:
Thank you for your contribution! I'll modify README after testing

fan-ziqi, 2024-10-08:
The installation in README has been updated!

closed 2024-10-08 by fan-ziqi