fan-ziqi/rl_sar #25: solution for segmentation fault

zdpisadaniu, 2024-10-10:
When i run the rl_sar program, it will crash in about 10 min. I found that it was caused by memory corruption using gdb .The 'robot state' was read by two thread at the same time.The solution of the problem is adding  two mutex lock .
for example
[image]
[image]
[image]

fan-ziqi, 2024-10-10:
Thank you for your correction. Can you provide a specific gdb debugging process or screenshot so that I can reproduce it on my computer

zdpisadaniu, 2024-10-10:
[image]
This is the gdb bt result,

fan-ziqi, 2024-10-16:
fixed in 26166fb

fan-ziqi, 2024-12-19:
Adding mutex will cause thread timeout, reopen this issue.

hotfixed in [44f8f53](https://github.com/fan-ziqi/rl_sar/commit/44f8f5324f55e07d7472a3744e2f02a39fff98c7)

fan-ziqi, 2025-01-11:
Fixed by using `tbb::concurrent_queue` instead of `std::mutex`
[4575a5b](https://github.com/fan-ziqi/rl_sar/commit/4575a5b62d1585eef6dfa34c7b892526540b0bbc)

closed 2025-01-11 by fan-ziqi