fan-ziqi/rl_sar #26: Bug in rl_sar During Real-world Deployment: Controller Crashes on Rotation Over 180 Degrees

zma69650, 2024-10-11:
When using the go2 functionality in real-world deployment, the controller crashes whenever the rotation exceeds 180 degrees.

fan-ziqi, 2024-10-11:
You can check if there is a jump in the quaternion at 180 degrees, preferably by providing screenshots or data
Also, is it convenient for you to contribute your go2 deployment code?
---- Replied Message ----
From JiJie ***@***.***> Date 10/11/2024 10:37 To ***@***.***> Cc ***@***.***> Subject [fan-ziqi/rl_sar] Bug in rl_sar During Real-world Deployment: Controller Crashes on Rotation Over 180 Degrees (Issue #26) 
When using the go2 functionality in real-world deployment, the controller crashes whenever the rotation exceeds 180 degrees.

zma69650, 2024-10-11:
rl_sar has been very helpful to me, thank you for your response!
I had the robot rotate counterclockwise through four orientations, then compared the quaternions between the simulation and the real-world deployment.
sim 1:
[image]
sim 2:
[image]
real 2:
[image]
sim 3:
[image]
real 3:
[image]
sim 4: 
[image]
real 4:
[image]
Could it be that the quaternion in the real-world deployment reached 1, causing the error? 
This could be a numerical issue in the code. Initially, the values were normal, but when I had go2 stand, the quaternion values only had two decimal places.

zma69650, 2024-10-11:
The quaternion received when it's functioning normally is:
[image]

zma69650, 2024-10-11:
前面搞错了，是打印的问题setprecision(2)，目前还没找到原因

zma69650, 2024-10-12:
在仿真中，当计算观察者中的ang_vel时，不使用四元数的话就会出现实机的情况：
```
obs_list.push_back(this->obs.ang_vel * this->params.ang_vel_scale);
```

https://github.com/user-attachments/assets/35d595aa-1cac-4318-83c2-6b70b3801b30

fan-ziqi, 2024-10-12:
我觉得是因为实机的角速度坐标系和仿真及训练不一样。

QuatRotateInverse函数的第一个参数是机器人角度四元数，第二个参数的坐标系是世界坐标系，函数的输出是第二个参数在机身坐标系下的值。

我认为在isaacgym及gazebo中，角速度的坐标系是世界坐标系，需要使用QuatRotateInverse进行坐标系转换，从而obs中角速度的坐标系为机身坐标系。

而在实机上，如果角速度的坐标系就是机身坐标系，则不需要进行转换。

如果在仿真中不使用QuatRotateInverse进行坐标系转换，则obs中的角速度坐标系就与实机一样了，从而导致与实机相同的问题。

你试一下实机运行的时候不适用QuatRotateInverse进行坐标系转换，看看能不能解决这个问题

zma69650, 2024-10-12:
你是对的，现在问题解决了，仿真中得加QuatRotateInverse，实机部署则不需要。我整理代码先，然后提交一下Go2的实机部署的代码。

fan-ziqi, 2024-10-12:
恭喜。这块坐标系的判定我也可能改一下，下周出差回去弄，改好了我在这个issue说一下
---- 回复的原邮件 ----
发件人 JiJie ***@***.***> 日期 2024年10月12日 12:41 收件人 ***@***.***> 抄送至 ***@***.***>***@***.***> 主题 Re: [fan-ziqi/rl_sar] Bug in rl_sar During Real-world Deployment: Controller Crashes on Rotation Over 180 Degrees (Issue #26) 
你是对的，现在问题解决了，仿真中得加QuatRotateInverse，实机部署则不需要。我整理代码先，然后提交一下Go2的实机部署的代码。

fan-ziqi, 2024-10-15:
#27

closed 2024-10-15 by fan-ziqi