fan-ziqi/rl_sar #30: [Bug Report] Bug title

jingdaliang, 2024-10-24:
source devel/setup.bash
(for python version) rosrun rl_sar rl_sim.py
在执行这一步的时候报错：Traceback (most recent call last):
  File "/home/robot/lite2/devel/lib/rl_sar/rl_sim.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/robot/lite2/src/rl_sar/src/rl_sar/scripts/rl_sim.py", line 17, in <module>
    from rl_sdk import *
ModuleNotFoundError: No module named 'rl_sdk'
如图
[image]
报错的位置如图
[image]
[image]

麻烦大佬解答

fan-ziqi, 2024-10-24:
我这边无法复现您的问题。

请根据下述模板详细说明你的软件版本以及你都做了什么操作，你都用了模板为什么要把下面的内容删掉呢？这对我帮你找到问题很有帮助。请直接修改你上面的内容，无需重新开一个新帖。

If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

A clear and concise description of what the bug is.

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- OS: [e.g. Ubuntu 20.04, Ubuntu 22.04, WSL2]
- ROS Version: [e.g. noetic, melodic]
- Python Version: [e.g. Python 3.8, Conda-Python 3.11]

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

jingdaliang, 2024-10-24:
解决啦，麻烦了，顺便问一下这个框架可以换任何四足机器人模型吗

fan-ziqi, 2024-10-24:
可以，任何机器人都可以

fan-ziqi, 2024-10-24:
请问你是如何解决的？可以给其他人一个参考吗

closed 2024-10-24 by fan-ziqi

jingdaliang, 2024-10-24:
就是路径问题，我本来是自己建了一个src把 rl_sar 放在里边，然后build的，现在直接在rl_sar这个根目录下build的，一开始理解错了，哈哈哈哈，麻烦大佬了