fan-ziqi/rl_sar #33: An error about undefined reference.

MaanEcho, 2024-10-27:
### Describe the bug

When I used the command 'catkin build' to build the project, an error occured. It seems that there exits a question in the linking phase, which is an error about undefined reference. 

### Steps to reproduce

catkin build

The error messages are shown in the picture below.

[image]

### System Info

- Commit: 1a92a2a
- OS: Ubuntu 20.04
- ROS Version: noetic
- Python Version: Python 3.8.10

### Additional context

None

jingdaliang, 2024-10-28:
楼主试一下这个export Torch_DIR=libtorch路径，然后重新编译，我一开始也有这个问题，就这样解决的

MaanEcho, 2024-10-28:
感谢回复！我之前已经在.bashrc中添加了export Torch_DIR，具体内容是export Torch_DIR=/home/fuhuangan/ThirdParty/libtorch/，添加之后已source。在你回复之后，我尝试新开终端，并在终端内直接执行export Torch_DIR=/home/fuhuangan/ThirdParty/libtorch/。然后在同一终端内，使用catkin clean指令清除之前build产生的build、devel和logs三个文件夹，并重新执行catkin build进行构建，但是还是报错，错误信息如图片所示。我使用的libtorch是原项目提供的指令下载的，应该不会有问题。

请问您还有什么好的建议供我参考吗？

heidisis, 2024-10-30:
可以试试在将cmakelist中的
add_executable(rl_sim src/rl_sim.cpp )
target_link_libraries(rl_sim
  ${catkin_LIBRARIES} -pthread
  rl_sdk observation_buffer yaml-cpp
)
修改成
add_executable(rl_sim src/rl_sim.cpp )
target_link_libraries(rl_sim
  ${catkin_LIBRARIES} -pthread
  rl_sdk observation_buffer yaml-cpp ${UUID_LIB}
)

heidisis, 2024-10-30:
@MaanEcho 可以试试在将cmakelist中的
add_executable(rl_sim src/rl_sim.cpp )
target_link_libraries(rl_sim
${catkin_LIBRARIES} -pthread
rl_sdk observation_buffer yaml-cpp
)
修改成
add_executable(rl_sim src/rl_sim.cpp )
target_link_libraries(rl_sim
${catkin_LIBRARIES} -pthread
rl_sdk observation_buffer yaml-cpp ${UUID_LIB}
)

MaanEcho, 2024-11-14:
I have found out the reason.
In the CMakeLists.txt in package rl_sar, there is one line 'link_directories(/usr/local/lib)'. So the build tool will try to find out all the libraries it needs in this path. But I also installed the unitree_sdk2 in this path in the past, which is not the same version compared with the one provided in this repository and there exists some differences in the api. I believe that the build tool used the version installed by myself, not the one provided by this repository, and this lead to the error. When I commented out this line, all the packages are built successfully!
Anyway, if anyone meets similar question, you can try to comment out this line, "link_directories(/usr/local/lib)"

fan-ziqi, 2024-11-14:
> I have found out the reason. In the CMakeLists.txt in package rl_sar, there is one line 'link_directories(/usr/local/lib)'. So the build tool will try to find out all the libraries it needs in this [...]

What you said seems to be correct. The `link_directories(/usr/local/lib)` is used to find the yaml-cpp and lcm libraries installed by apt. Users can modify it as needed.

closed 2024-11-14 by fan-ziqi