fan-ziqi/rl_sar #39: [Bug Report] ROS2 Gazebo Simulation Physics Issue

T-K-233, 2024-11-10:
### Describe the bug

When running in ROS2 environment, the robot will suddenly jump up after a few seconds.

In simulation, the robot dog is able to walk a few steps, but then will suddenly jump up (around timestamp 00:20), as shown in the video. This is observed in multiple simulations under different commands.

https://github.com/user-attachments/assets/349348df-c8bd-47cc-af72-a54eb1a0b293

I don't think the ROS1 version has the same issue.

### Steps to reproduce

```bash
ros2 launch rl_sar gazebo.launch.py rname:=a1 framework:=isaacgym
```

and then

```bash
ros2 run rl_sar rl_sim.py
```

### System Info

System environment:

- Commit: [64fc72d](https://github.com/fan-ziqi/rl_sar/tree/64fc72d613425b332a002bb62be7956f43361db3)
- OS: Ubuntu 20.04
- ROS Version: ROS2 Foxy
- Python Version: Python 3.8 (from Ubuntu system)

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

fan-ziqi, 2024-11-12:
It seems that there is an error in the angle quaternion. I will check this, thank you for your feedback!

fan-ziqi, 2024-11-13:
fixed in [4124f78](https://github.com/fan-ziqi/rl_sar/commit/4124f78e1cae2d850992dfa3aa7595cbef936ab2)

commit 4124f78e (2024-11-13): fix: the angular velocity of the IMU topic is in the body coordinate system. Close #39