fan-ziqi/rl_sar #42: [Question] Deploy pt model on GO2.

zhangOSK, 2024-12-03:
### Question

 你好，很实用的仓库，已star。请问怎么在go2部署zeonsunlightyu/LocomotionWithNP3O？
 我看到你b站视频里已经跑过了。 我不太清楚config.yaml文件应该怎么配置？能多给点信息吗，谢谢。

fan-ziqi, 2024-12-03:
应该差距不大吧，`config.yaml`里面的参数基本都是RL训练的参数

kong3166, 2024-12-04:
你好，很实用的仓库，已star。请问你这个仓库中go2的模型himloco.pt 是通过HIMLoco这个强化学习项目训练出来的么？还是通过哪个项目训练的？我这两天用HIMLoco这个项目训练了一些go2的模型 但是并不能用rl_sar加载这些模型，非常期待回复，谢谢。

fan-ziqi, 2024-12-04:
是的，请问遇到什么问题了？
---- 回复的原邮件 ----
发件人 ***@***.***> 日期 2024年12月04日 10:08 收件人 ***@***.***> 抄送至 ***@***.***>***@***.***> 主题 Re: [fan-ziqi/rl_sar] [Question] Deploy pt model on GO2. (Issue #42) 
你好，很实用的仓库，已star。请问你这个仓库中go2的模型himloco.pt 是通过HIMLoco这个强化学习项目训练出来的么？还是通过哪个项目训练的？我这两天用HIMLoco这个项目训练了一些go2的模型 但是并不能用rl_sar加载这些模型，非常期待回复，谢谢。

kong3166, 2024-12-04:
非常感谢回复，
问题出现在第2步启动rosrun rl_sar rl_sim，在加载模型处报错，模型路径我打印出来了，查看后模型路径没问题，但是加载模型就会报错，后来我将路径换成模型的绝对路径，依然会报错。
错误输出：terminate called after throwing an instance of ‘c10::Error’

发自我的iPhone

 发件人 ***@***.***&gt; 日期 2024年12月04日 10:08 收件人 ***@***.***&gt; 抄送至 ***@***.***&gt;***@***.***&gt; 主题 Re: [fan-ziqi/rl_sar] [Question] Deploy pt model on GO2. (Issue #42)

fan-ziqi, 2024-12-04:
请贴一下完整的报错内容

kong3166, 2024-12-05:
感谢回复
运行命令rosrun rl_sar rl_sim时，报错如下图：

然后我把rl_sim.py中的torch.jit.load() 改成了torch.load()后，运行命令rosrun rl_sar rl_sim.py，报错如下图：

fan-ziqi, 2024-12-05:
图没了

kong3166, 2024-12-05:
嘿嘿　我已经找到原因了大佬　现在可以成功加载训练好的策略了　很感谢

fan-ziqi, 2024-12-05:
不用谢。

closed 2024-12-05 by fan-ziqi