fan-ziqi/rl_sar #44: [Question] sim2sim:rl_sim.py

Benxiaogu, 2025-01-11:
### Question

I trained a a1 model by legged_gym, and I tried sim2sim by rl_sar. I met a error whe loading the model, the robot in gazebo doesn't move when i pressed the keys w, s,...
[image]
I do not know how to solve it, i will preciate it if you could give an answer

fan-ziqi, 2025-01-11:
Your dimensions do not match. Please check carefully

closed 2025-01-11 by fan-ziqi

Benxiaogu, 2025-01-11:
> Your dimensions do not match. Please check carefully

Yeah, I know that my dimensions have mismatch. The matter is that I really don't know how to check and fix it. Could you give me some guide please?

homemoon-cpu, 2025-02-07:
> > Your dimensions do not match. Please check carefully
> 
> Yeah, I know that my dimensions have mismatch. The matter is that I really don't know how to check and fix it. Could you give me some guide please?

Hello!Have you solved this problem now?I meet it,too.