fan-ziqi/rl_sar #45: [Question] ROS Humble Support

JakeFishcode, 2025-01-15:
### Question
```
/usr/bin/ld: CMakeFiles/rl_real_go2.dir/src/rl_real_go2.cpp.o: in function `RL_Real::RL_Real()':
/opt/rl_sar/src/rl_sar/src/rl_real_go2.cpp:25: undefined reference to `unitree::robot::go2::RobotStateClient::ServiceSwitch(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/rl_real_go2.dir/build.make:104: rl_real_go2] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:252: CMakeFiles/rl_real_go2.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< rl_sar [25.0s, exited with code 2]

Summary: 4 packages finished [33.6s]
  1 package failed: rl_sar
  2 packages had stderr output: rl_sar robot_joint_controller
```
In Ubuntu22.04 ros humble
It this is the reason that only support foxy?

fan-ziqi, 2025-01-18:
This project does support ros2-humble, It looks like a problem with the go2 SDK.

fan-ziqi, 2025-02-12:
The issue is assumed to be fixed. Please reopen in case problems still exist.

closed 2025-02-12 by fan-ziqi