fan-ziqi/rl_sar #46: [Question] How to train a new robot?

Feedstar, 2025-02-08:
### Question

大佬你好。我有看到你的go2用的pt文件前缀是himloco，应该用的是这个项目训练出来的模型。我在使用himloco训练自己的机器狗时，sim2sim后的效果很差，但你的模型鲁棒性似乎很好，甚至你的go2的pt文件直接用在我的机器狗上时效果也不错。所以我想请问方便发布你的himloco训练过程吗。

我使用的sim2sim框架是[quadruped_ros2_control](https://github.com/legubiao/quadruped_ros2_control)

fan-ziqi, 2025-02-08:
这个策略是我一个朋友训练的，我叫上他来一起讨论

zma69650, 2025-02-08:
hello @Feedstar ,你可以先在legged gym播放一下你训练的策略或者查看训练的数据，先保证训练的策略是没有问题。如果这没有问题，再检查sim2sim。不过你提到用rl_sar提供的pt文件是可以，那估计是你训练的问题。关于训练，我没有做改动，只是增加了go2模型，参数都没有改动，用的是a1的参数。

Feedstar, 2025-02-10:
> hello [@Feedstar](https://github.com/Feedstar) ,你可以先在legged gym播放一下你训练的策略或者查看训练的数据，先保证训练的策略是没有问题。如果这没有问题，再检查sim2sim。不过你提到用rl_sar提供的pt文件是可以，那估计是你训练的问题。关于训练，我没有做改动，只是增加了go2模型，参数都没有改动，用的是a1的参数。

感谢回复。我在gym中play的时候机器狗表现正常，只在mujoco迁移时表现糟糕。训练参数我也照搬了a1。

https://github.com/user-attachments/assets/eb8f845a-91b2-4093-b944-283daa869cdf

https://github.com/user-attachments/assets/009e8dc4-4402-48da-a23d-53d7807788ce

zma69650, 2025-02-11:
机器人看起来像是你们自己的狗子，这样的话一些参数可能得做一些调整，比如kp,kd，为了更好的迁移到mujoco，还可以调整一下域随机化

Feedstar, 2025-02-12:
这样看来我的机器狗和himloco原生的参数非常不匹配，后续我会不断进行调整以让mujoco中的效果逼近isaac gym。再次感谢你们的工作与回复！

fan-ziqi, 2025-02-12:
感谢 @zma69650 的解答。关闭此issue，如有问题欢迎随时reopen！

closed 2025-02-12 by fan-ziqi