fan-ziqi/rl_sar #55: [Bug Report] Go2 Onboard Jetson Troubleshoot

Charlescai123, 2025-03-23:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Thanks for the nice opensource project. Actually I ran into some troubles after deploying this framework to Go2's onboard Jetson NX. I used ROS2 version foxy and the compilation runs smoothly by switching the x86_64 dependency to aarch64. However, when I began running the program, I encountered a crash shortly after sending a few teleoperation commands. The terminal reported the following error: **what(): cannot create std::vector larger than max_size()** and the program was terminated. See the screenshot below:

### Steps to reproduce

1. Terminal run: 
```
ros2 run rl_sar rl_real_go2 eth0 
```

2. Press **R1** and teleoperate the robot for a short time

### System Info

Describe the characteristic of your environment:

- Commit: 66c41
- OS: Ubuntu 20.04
- ROS Version: foxy
- Python Version: Python3.10

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

zhangfanyuanhang9, 2025-05-19:
Hello, I also tried to deploy this project to an ARM platform, but I encountered some problems. Would you mind sharing how to switch the x86_64 dependencies to aarch64? Thank you!

Charlescai123, 2025-05-19:
> Hello, I also tried to deploy this project to an ARM platform, but I encountered some problems. Would you mind sharing how to switch the x86_64 dependencies to aarch64? Thank you!

You can follow the tutorials there: https://blog.csdn.net/littlewells/article/details/146235689?spm=1001.2014.3001.5501

fan-ziqi, 2025-06-08:
我在编译的时候遇到了如下问题：

```bash
Errors     << rl_sar:cmake /home/unitree/rl_sar/logs/rl_sar/build.cmake.000.log                                                                                                                                                     
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "controller_manager" with any of the following names:

    controller_managerConfig.cmake
    controller_manager-config.cmake

  Add the installation prefix of "controller_manager" to CMAKE_PREFIX_PATH or
  set "controller_manager_DIR" to a directory containing one of the above
  files.  If "controller_manager" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:19 (find_package)
```
如果我手动指定路径 `list(APPEND CMAKE_PREFIX_PATH "/opt/ros/noetic")`，又会报错：

```bash
Errors     << rl_sar:cmake /home/unitree/rl_sar/logs/rl_sar/build.cmake.003.log                                                                                                                                                     
WARNING: package 'gazebo_ros' is deprecated (This package has been deprecated as of January 2025 with Gazebo classic 11 reaching
      end-of-life. Users are highly encouraged to migrate to the new Gazebo
      using our migration guides (https://gazebosim.org/docs/latest/gazebo_classic_migration))
WARNING: package 'gazebo_msgs' is deprecated (This package has been deprecated as of January 2025 with Gazebo classic 11 reaching
      end-of-life. Users are highly encouraged to migrate to the new Gazebo
      using our migration guides (https://gazebosim.org/docs/latest/gazebo_classic_migration))
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "robot_msgs" with
  any of the following names:

    robot_msgsConfig.cmake
    robot_msgs-config.cmake

  Add the installation prefix of "robot_msgs" to CMAKE_PREFIX_PATH or set
  "robot_msgs_DIR" to a directory containing one of the above files.  If
  "robot_msgs" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:20 (find_package)

```

系统的ros有重装过，不知道你又没有遇到过这个问题？ @Charlescai123

kunen23, 2025-06-08:
You just need to modify the CMakeLists.txt in rl_sar  to fix the issue.
`find_package(catkin REQUIRED COMPONENTS
  controller_manager
  genmsg
  joint_state_controller
  robot_state_publisher
  roscpp
  gazebo_ros
  std_msgs
  tf
  geometry_msgs
  robot_msgs
  robot_joint_controller
  rospy
)

find_package(Python3 COMPONENTS Interpreter Development REQUIRED)

catkin_package(
  CATKIN_DEPENDS
  robot_joint_controller
  rospy
)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${TORCH_CXX_FLAGS}")
find_package(Torch REQUIRED)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${GAZEBO_CXX_FLAGS}")
find_package(gazebo REQUIRED)

find_package(TBB REQUIRED)
`
@fan-ziqi

Charlescai123, 2025-06-10:
> 我在编译的时候遇到了如下问题：
> 
> Errors     << rl_sar:cmake /home/unitree/rl_sar/logs/rl_sar/build.cmake.000.log                                                                                                                                                     
> CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
>   Could not find a package configuration file provided by
>   "controller_manager" with any of the following names:
> 
>     controller_managerConfig.cmake
>     controller_manager-config.cmake
> 
>   Add the installation prefix of "controller_manager" to CMAKE_PREFIX_PATH or
>   set "controller_manager_DIR" to a directory containing one of the above
>   files.  If "controller_manager" provides a separate development package or
>   SDK, be sure it has been installed.
> Call Stack (most recent call first):
>   CMakeLists.txt:19 (find_package)
> 如果我手动指定路径 `list(APPEND CMAKE_PREFIX_PATH "/opt/ros/noetic")`，又会报错：
> 
> Errors     << rl_sar:cmake /home/unitree/rl_sar/logs/rl_sar/build.cmake.003.log                                                                                                                                                     
> WARNING: package 'gazebo_ros' is deprecated (This package has been deprecated as of January 2025 with Gazebo classic 11 reaching
>       end-of-life. Users are highly encouraged to migrate to the new Gazebo
>       using our migration guides (https://gazebosim.org/docs/latest/gazebo_classic_migration))
> WARNING: package 'gazebo_msgs' is deprecated (This package has been deprecated as of January 2025 with Gazebo classic 11 reaching
>       end-of-life. Users are highly encouraged to migrate to the new Gazebo
>       using our migration guides (https://gazebosim.org/docs/latest/gazebo_classic_migration))
> CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
>   Could not find a package configuration file provided by "robot_msgs" with
>   any of the following names:
> 
>     robot_msgsConfig.cmake
>     robot_msgs-config.cmake
> 
>   Add the installation prefix of "robot_msgs" to CMAKE_PREFIX_PATH or set
>   "robot_msgs_DIR" to a directory containing one of the above files.  If
>   "robot_msgs" provides a separate development package or SDK, be sure it has
>   been installed.
> Call Stack (most recent call first):
>   CMakeLists.txt:20 (find_package)
> 系统的ros有重装过，不知道你又没有遇到过这个问题？ [@Charlescai123](https://github.com/Charlescai123)

哦我这里测试的时候用的是ros2分支，拓展坞上装的是宇树原版的系统镜像，编译时终端那边弹出来的foxy和noetic要选择foxy应该才能设置对应的ros2环境。我看报错应该是选择了ros1的缘故，重新编译环境再改成ros2的试试？

Charlescai123, 2025-06-14:
好的 感谢范总