fan-ziqi/rl_sar #59: [Question] undefined reference to `unitree::robot::go2::RobotStateClient::ServiceSwitch

TianqiYu123, 2025-04-04:
### Question

Thanks for your work! I was trying to build this project on Ubuntu 20.04 with ROS noetic but received this error when I followed the building procedure. Do you have any idea how to fix this? Thx

...................................................................................................................
___________________________________________________________________________________________________________________
Errors     << rl_sar:make /home/rl/rl_sar/logs/rl_sar/build.make.000.log                                        
/usr/bin/ld: CMakeFiles/rl_real_go2.dir/src/rl_real_go2.cpp.o: in function `RL_Real::RL_Real()':
/home/rl/rl_sar/src/rl_sar/src/rl_real_go2.cpp:32: undefined reference to `unitree::robot::go2::RobotStateClient::ServiceSwitch(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rl_real_go2.dir/build.make:140：/home/rl/rl_sar/devel/.private/rl_sar/lib/rl_sar/rl_real_go2] 错误 1
make[1]: *** [CMakeFiles/Makefile2:1635：CMakeFiles/rl_real_go2.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
cd /home/rl/rl_sar/build/rl_sar; catkin build --get-env rl_sar | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...................................................................................................................
Failed     << rl_sar:make                           [ Exited with code 2 ]                                         
Failed    <<< rl_sar                                [ 28.5 seconds ]                                               
[build] Summary: 10 of 11 packages succeeded.                                                                      
[build]   Ignored:   None.                                                                                         
[build]   Warnings:  1 packages succeeded with warnings.                                                           
[build]   Abandoned: None.                                                                                         
[build]   Failed:    1 packages failed.                                                                            
[build] Runtime: 29.9 seconds total.

yitingHH-bit, 2025-05-10:
how you solved it

yitingHH-bit, 2025-05-10:
https://github.com/fan-ziqi/rl_sar/issues/47

fan-ziqi, 2025-06-12:
Please try the latest code