fan-ziqi/rl_sar #63: [Question] In isaacsim framework, robot shaking

sakamotor223, 2025-05-08:
Thanks for the open source. I found in my test that himloco performs better than the policy under the isaacsim framework. The robot under the isaacsim framework has joint shaking. The policy that comes with the code and the policy under the isaaclab that I trained have this problem. I really don't find the problem.

https://github.com/user-attachments/assets/7bb75abb-cebb-4aa1-b367-16ebe6488359

sakamotor223, 2025-05-08:
The policy I trained myself performed well in the lab, but after deployment it also jittered.
https://github.com/user-attachments/assets/e2067cdf-6461-4736-aa07-cd24b825f7df

waynechu1109, 2025-06-09:
I got the same problem. Do @fan-ziqi have any suggestion on this issue?
Is using v2.3 of this repo helpful?

waynechu1109, 2025-06-09:
@sakamotor223  decrease the rl_kp and increase rl_kd will improve performance

xiaoxiaconfidence, 2025-07-15:
Hi @sakamotor223 ,
I hope you're doing well! I noticed this issue is still open and wanted to check if you've managed to resolve the robot shaking problem in IsaacSim framework?
I'm experiencing a very similar issue during sim2sim transfer to Gazebo with a heavy quadruped robot (90kg).

abizovnuralem, 2025-07-22:
> [@abizovnuralem](https://github.com/abizovnuralem) Thanks. Do we need to change the PD operation in sim2real?

No need, you can use the author's PD