fan-ziqi/rl_sar #113: The robot cannot keep GetUp and will automatically GetDown after GetUp completely.then, up and down repeatedly

N0rma1zz, 2025-08-22:
### Description

When pressing '0' (Keyboard),the robot will automatically GetDown after GetUp 100%,and switch to GetDown 0.4%.Then, up and down like that repeatedly.
I just ran the code without making any changes and all robots behave like that.
[image]
### Steps to reproduce
```
source devel/setup.bash
roslaunch rl_sar gazebo.launch rname:=go2
```
and then
```
source devel/setup.bash
rosrun rl_sar rl_sim
```

### Commit

e373b94

### Operating System (OS)

Ubuntu 20.04

### ROS Version

ROS Melodic

### Python Version

Python 3.8

zzzltmy, 2025-08-27:
我在a1上没有出现这个问题，在尝试go2w的时候也会出现这个问题

Ma-Puzhi, 2025-08-27:
I meet the same error 
Operating System (OS): ubuntu20.04
ROS: noetic
I only run: 
> ./build.sh
> source devel/setup.bash
> roslaunch rl_sar gazebo.launch rname:=go2
> open new terminal
> rosrun rl_sar rl_sim

fan-ziqi, 2025-08-27:
That is weird, I can't reproduce this issue

N0rma1zz, 2025-08-27:
oh,the ROS version is noetic.my another pc can easily reproduce this issue with the same OS and ROS version

fjc378209410, 2025-08-29:
I have the same problem

fjc378209410, 2025-08-29:
> my environment is Ubuntu 20.04 + ROS1 Noetic我的环境是Ubuntu 20.04 ROS1 Noetic

May I ask if the keyboard can control the movement of go2 in v3.0.1? I checked that the go2 *. urdf version v3.0.1 is not the official latest version, and the latest version should have 1217 lines.

dengjieqwer, 2025-08-29:
> > my environment is Ubuntu 20.04 + ROS1 Noetic我的环境是Ubuntu 20.04 ROS1 Noetic
> 
> May I ask if the keyboard can control the movement of go2 in v3.0.1? I checked that the go2 *. urdf version v3.0.1 is not the official latest version, and the latest version should have 1217 lines.

3.0.1 version can be keyboard controlled, I can't use the latest version

fjc378209410, 2025-08-29:
> my environment is Ubuntu 20.04 + ROS1 Noetic

I changed the code of v3.0.1 and found a bug in the priority of Joystick and Keboard. After the modification, keyboard control can now be implemented. I am currently trying to deploy my trained model to Gazebo for testing

commit 7611d041 (2025-09-17): fix: FSM repeatedly toggling between RLFSMStateGetDown and RLFSMStateGetUp (#124)

Fixes #113
closed 2025-09-17 by fan-ziqi