fan-ziqi/rl_sar #115: G1 humanoid robot fails to stand upright - lies on side during GetUp sequence

linjiw, 2025-08-25:
### Description

## Issue Description

### Problem Summary
The G1 humanoid robot consistently fails to achieve a standing position during the GetUp finite state machine (FSM) sequence. Instead of standing upright, the robot lies on its side throughout the entire process.

### Environment
- **Platform**: Ubuntu 22.04 LTS
- **ROS2 Version**: Humble
- **Simulation**: Gazebo
- **Robot**: G1 humanoid
- **Branch**: main (commit ad19ccc)

### Expected Behavior
When pressing `Num0` to trigger the GetUp sequence, the G1 robot should:
1. Transition from initial pose to a stable standing position
2. Maintain upright orientation (Roll ≈ 0°, Pitch ≈ 0°)
3. Complete GetUp sequence successfully at 100%

### Actual Behavior
The G1 robot:
1. Starts lying on its side (Roll ≈ -109°)
2. Remains on its side throughout the GetUp sequence
3. Completes at 100% but still lying sideways (Roll ≈ -90°)
4. Never achieves upright standing posture

### Debug Analysis - Key Findings

I've implemented comprehensive debug logging and analyzed the GetUp sequence. Here are the critical findings:

#### 1. IMU Orientation Data
```
Progress 0%:   Roll=-109.2° Pitch=44.5°  Yaw=0.0°   (lying on side)
Progress 20%:  Roll=-101.3° Pitch=35.2°  Yaw=0.0°   (still sideways)
Progress 40%:  Roll=-95.7°  Pitch=22.1°  Yaw=0.0°   (still sideways)
Progress 60%:  Roll=-92.4°  Pitch=15.3°  Yaw=0.0°   (still sideways)
Progress 80%:  Roll=-90.8°  Pitch=8.7°   Yaw=0.0°   (still sideways)
Progress 100%: Roll=-89.9°  Pitch=5.2°   Yaw=0.0°   (FAILED - still sideways)
```

**Expected**: Roll ≈ 0° for upright standing
**Actual**: Roll ≈ -90° (robot lying on left side)

#### 2. Joint Position Analysis
**Target positions** (from default_dof_pos in base.yaml):
```yaml
default_dof_pos: [-0.1, 0.0, 0.0, 0.3, -0.2, 0.0,    # Left leg
                  -0.1, 0.0, 0.0, 0.3, -0.2, 0.0,    # Right leg  
                  0.0, 0.0, 0.0,                     # Waist
                  0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, # Left arm
                  0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0] # Right arm
```

**Current positions** track targets accurately, indicating the control system works but the target pose itself is problematic.

#### 3. Root Cause Hypothesis
The default joint configuration in `src/rl_sar/policy/g1/base.yaml` appears to define an **inherently unstable pose** that results in the robot falling to its side rather than maintaining an upright standing position.

### Attempted Solutions

1. **✅ Debug Logging Implementation**: Added comprehensive FSM debug logging to track joint positions, IMU data, and control parameters
2. **❌ Default Pose Correction**: Attempted to implement a more stable humanoid standing pose with:
   - Deeper leg squat for lower center of gravity
   - Wider stance for lateral stability
   - Slight forward lean for balance
3. **❌ Result**: Robot behavior unchanged, still lies on side

### Technical Details

#### Debug Implementation
- Added detailed logging to `src/rl_sar/policy/g1/fsm.hpp` GetUp state
- Captures joint positions, IMU orientation, control gains every 20% of progress
- Full debug logs available in attached files

#### Key Files Involved
- `src/rl_sar/policy/g1/base.yaml` - Robot configuration and default poses
- `src/rl_sar/policy/g1/fsm.hpp` - Finite State Machine implementation
- `src/rl_sar/policy/g1/` - G1-specific policy implementation

### Attachments
[image]
- [ ] Screenshot of robot lying on side in [Gazebo]([url](url))

### Possible Investigation Directions

1. **Physics/URDF Issues**: Robot model or physics parameters causing instability
2. **Joint Mapping**: Potential mismatch between joint indices and actual robot joints
3. **Control Parameters**: Inappropriate kp/kd gains for G1 humanoid kinematics
4. **FSM Logic**: GetUp sequence logic may be designed for different robot type
5. **Initial Conditions**: Simulation starting conditions incompatible with GetUp FSM

### Environment Setup for Reproduction

```bash
# Build project
source /opt/ros/humble/setup.bash
./build.sh rl_sar

# Terminal 1: Start G1 simulation
ros2 launch rl_sar gazebo.launch.py rname:=g1

# Terminal 2: Run with debug logging  
ros2 run rl_sar rl_sim 2>&1 | tee reproduction_log.txt

# Trigger issue: Press Num0 in rl_sim terminal
```

### Request for Help

This appears to be a fundamental issue with the G1 robot configuration or FSM logic. The debug data clearly shows the problem but the underlying cause requires deeper investigation into:

- G1 robot model and kinematics
- GetUp FSM design assumptions
- Default pose stability analysis
- Physics simulation parameters

Any insights into G1-specific requirements or known issues would be greatly appreciated!

---

**Labels**: `bug`, `g1`, `simulation`, `fsm`, `help-wanted` 

### Commit

main

### Operating System (OS)

Ubuntu 22.04

### ROS Version

Humble

### Python Version

Python 3.10 

### Additional information

[image]

fan-ziqi, 2025-08-26:
Thanks for reporting, I will check this.

IgararA, 2026-01-16:
> I can successfully deploy on Ubuntu 20.04

Could you share how you achieved this? I am running the same environment (**Ubuntu 20.04 + ROS Noetic**), but I'm encountering issues with the `G1` robot.

**Steps to Reproduce:**
1. Run `rosrun rl_sar rl_sim`
2. Press `Num0`: The robot stands up successfully.
3. Press `R`: The robot resets to the initial position (this takes a significantly long time).
4. Press `Num0` or `Num1` immediately after the reset: **The robot fails to stand up again.**

**Questions:**
- Is there something wrong with my command?
- What is the current support status for G1? I checked the discussions but it's unclear.
*Note: I have observed this behavior in both `v4.0.0` and `v3.1.0`.*