fan-ziqi/rl_sar #118: The index of the tau_est in robot_joint_controller_group.cpp

whwdadada, 2025-09-01:
### Description

The function **UpdateFunc** int the robot_joint_controller_group.cpp, the currentPos of the joint is obtained sa follows
https://github.com/fan-ziqi/rl_sar/blob/0503f4e3101fbdb98f6a42de4793ed6e5a2b53f5/src/robot_joint_controller/ros2/src/robot_joint_controller_group.cpp#L262
And the tau_est is get 
https://github.com/fan-ziqi/rl_sar/blob/0503f4e3101fbdb98f6a42de4793ed6e5a2b53f5/src/robot_joint_controller/ros2/src/robot_joint_controller_group.cpp#L271
I guess the correct index should be
last_state_.motor_state[index].tau_est = state_interfaces_[**2 * index + 1**].get_value();
not
last_state_.motor_state[index].tau_est = state_interfaces_[**index + 1**].get_value();

### Commit

main

### Operating System (OS)

Ubuntu22.04

### ROS Version

humble

### Python Version

python3.8

fan-ziqi, 2025-09-04:
Thanks for the feedback, I will check this.

fan-ziqi, 2025-09-04:
Thanks for pointing this! Fixed in [4c9b9b0](https://github.com/fan-ziqi/rl_sar/commit/4c9b9b0bf5897954e9f33c1dfddf4f56c384695c)

closed 2025-09-04 by fan-ziqi